gdombiak/OctoPod #616: iOS 16 App Widget

tempus2016, 2022-10-09:
I’m finding that the widget on iOS 16 won’t display any information if I choose a camera in the config. I can choose the printer & theme and it works, but if I choose a camera, the widget stops working. 

[image]
[image]
[image]

ah03, 2022-10-29:
I have the same issue

Arketsu, 2023-01-11:
Same issue for months almost a whole year.

Arketsu, 2023-01-23:
@gdombiak  just noticed that when using ip webcam from Pavel Khlebovich (used to work) camera won’t show in the widget. When using droidcamx from Dev47Apps (installed recently) widget works.

gdombiak, 2023-01-23:
Hey guys,

I've been trying to reproduce this locally without much success. The only scenario where I see the widget showing information but no camera image is this one:

1. Phone connected to local network
2. OctoPrint is available in local network 
3. OctoPrint is available via port forwarding protected with user/password
4. Cameras configured in OctoPrint with local network work fine
5. Cameras configured in OctoPrint with public URL DO NOT WORK since authentication fails (see step 1)

Is this your setup? If not I need more information to know how to reproduce what you are seeing. Please include camera resolution since that could be another reason.

Thanks,
Gaston

Arketsu, 2023-01-23:
> Hey guys,
>
> I've been trying to reproduce this locally without much success. The only scenario where I see the widget showing information but no camera image is this one:
>
> 1. Phone connected to local [...]

1. Yes
2. Local and public
3. Yes
4. Public tried local but octoprint won’t show local camera only public
5. Using direct video feed with no authentication needed via public ip

Morcegolas, 2023-01-28:
> 

1. Yes
2. 

> Hey guys,
>
> I've been trying to reproduce this locally without much success. The only scenario where I see the widget showing information but no camera image is this one:
>
> 1. Phone connected to local [...]

1. Yes
2. Yes
3. No
4. Yes
5. Using a VPN

Widget camera is not working for some time now.

Arketsu, 2023-03-01:
@tempus2016  @Morcegolas  are you guys running octoprint on windows or raspberry pi?

Morcegolas, 2023-03-01:
> @tempus2016 @Morcegolas are you guys running octoprint on windows or raspberry pi?

Raspberry Pi here

Arketsu, 2023-03-04:
Windows here, currently change the webcam to a Wyze cam and hosting the video/webcam server in the same windows environment. Still having the same issue with no video or job status on the widget only able to see if I open the app.

Ps: For some odd reason video fee inside the app only work with public IP and not local IP.

Arketsu, 2023-03-26:
@gdombiak new version of the plug-in might’ve fixed the issue atm still testing I will let you know this week if it’s 100% working.

Arketsu, 2023-04-07:
@gdombiak even after the last update for the plugin and app still doing the same thing.
@Morcegolas  @tempus2016  @ah03 still having the same problem after updating?

Morcegolas, 2023-04-07:
> @gdombiak even after the last update for the plugin and app still doing the same thing.
> @Morcegolas @tempus2016 @ah03 still having the same problem after updating?

Here everything is working great quite some time.

Arketsu, 2023-04-09:
@gdombiak 
Ended up creating a whole new test server with fresh octoprint installation and configuration latest octopod plugin and iOS app. Used a new iPhone 14 pro max and still have the same issue where widget does not show camera. Not sure if there any other troubleshooting step I can do since ended up doing a fresh install and nothing.

johndelfino, 2023-04-29:
> Hey guys,
>
>
>
> I've been trying to reproduce this locally without much success. The only scenario where I see the widget showing information but no camera image is this one:
>
>
>
> 1. Phone connected to [...]

I have been experiencing this issue since installing the application. 

1. Yes
2. Yes, local and public
3. Yes
4. Yes
5. No public URL

What's extra weird is that if assigning a camera to a medium widget breaks both that widget and a a separate small widget with no camera. It breaks both instances. Really bizarre stuff.

Morcegolas, 2023-05-23:
Today I changed my camera resolution for 1080p, the widget preview stopped working, if I change it back for 720p it's working again.

Arketsu, 2023-06-01:
Confirmed changing the resolution works…. @gdombiak can you make it so it supports other resolutions?

Morcegolas, 2023-06-01:
> Confirmed changing the resolution works…. @gdombiak can you make it so it supports other resolutions?

Glad I could help 👌🏼