gdombiak/OctoPod #617: Maintain printer list sort order when restoring on new device from iCloud

tideline3d, 2022-10-10:
I have a few iPads I'm deploying for the print farm. They're on the same iCloud account but when I restore the printers on the new devices, they're in a seemingly random order.  

It'd be nice if the sort order was stored with the iCloud backup so that the printer list and dashboard order are the same across devices.  I'd personally even be OK if I could just sort the list by name as my printers are named A1, A2, ... C4, C5 etc denoting their shelf(letter) and location(#). But I suspect others have a specific order that wouldn't text sort properly. 

I have only 24 printers to re-sort across 3 devices, I'll get over it for the initial setup and just manually fix them.  Thought I'd write this down for future me & others.

gdombiak, 2022-10-11:
Hi @tideline3d ,

When I implemented iCloud sync I knew that some day the ask to also store order of printers was going to come. :) I scheduled for next release this enhancement. At the very least, I will include sort by name in the next release.

Thanks,
Gaston

closed 2022-11-27 by gdombiak

tideline3d, 2022-12-09:
This worked perfectly across 3 devices on the latest store release. Thanks 👍

commit b7d0b8a8 (2022-11-27): Store printer position in iCloud

Closes #617