gdombiak/OctoPod #618: Print Job won't show details

Sparksaurus, 2022-10-14:
I connected to Octopod using octoeverywhere, in widget i get progress percentage but doesn't refresh to the right percentage, but it shows percentage and estimated time when it will finish by. But once i open the app it shows no information, I've attached image of what i'm seeing on my phone once the print starts

[image]

Drudoo, 2023-02-17:
I have the same issue. Both widgets and Apple Watch shows progress but the app itself doesn’t. 

I’m connecting to octoprint using reverse proxy.

Drudoo, 2023-02-23:
The issue seems to be fixed if you update your printer firmware. I updated to the latest stock firmware on my Genius Pro and the progress in the app works now.

Trilis29, 2023-05-10:
same for me, seems something wrong with reverse proxy... can't find any info