gdombiak/OctoPod #620: Octopod not showing thumbnail

megatech1966, 2022-10-26:
I just updated to Marlin 2.1.1 and octopod no longer shows the thumbnails on new gcode. OctoPrint shows them fine.

joshuacant, 2022-10-27:
I'm seeing this too, and I haven't made any changes to my setup. Thumbnails were working as recently as just a couple days ago, but every file I've sliced in PrusaSlicer (haven't changed the version) since then has only shown the generic icon. All the older files show thumbnails just fine. It's only new files I send to Octoprint that don't display in Octopod. They do appear just fine in the Octoprint UI.

Not sure what else to test, but happy to do any troubleshooting steps to figure it out.

closed 2022-11-24 by gdombiak

megatech1966, 2022-11-27:
This issue is closed. But I see no update. Is there a update coming?

gdombiak, 2022-11-27:
yep, it's fixed and ready for next release.  Want to fix a couple of more things and then push new release out.

commit 8fa3121d (2022-11-24): Plugin provides encoded URL so we need to decode it here. It will be encoded again when image is requested

Closes #620