gdombiak/OctoPod #625: Live activities iOS

Morcegolas, 2022-12-21:
Hello, is there any chance to add this wonderful app to the live activities, it would be awesome.

josephsamela, 2023-01-28:
I second this request! Adding a live activity notification would be super helpful. Here's a concept of what an Octoprint live activity notification might look like. The idea here is that during a print the live activity notification would appear showing live information such as... 
- Printer name
- File name
- Tool temperature
- Bed temperature
- Percent complete
- Est. time remaining
- Webcam preview

[image]

gdombiak, 2023-03-26:
dup of https://github.com/gdombiak/OctoPod/issues/635

closed 2023-03-26 by gdombiak