gdombiak/OctoPod #628: Login?

OrthoSteveK, 2023-01-17:
How do I access my Octoprint account from Octopod?  I cannot find a <login> or <connect> button or anywhere to enter my ID and password.

gdombiak, 2023-01-17:
Hi @OrthoSteveK ,

which option are you using when adding a new printer? The easiest is using Global API keys. You don’t have to enter your user/password of OctoPrint. Instead you use the global api key of OctoPrint. The user and password that you see in OctoPod is only used when you have a reverse proxy in front of your OctoPrint when that’s used when trying to access your printers when outside of your home. 

Some old but still relevant documentation can be found here https://github.com/gdombiak/OctoPod/wiki/How-to-add-new-printers-to-OctoPod%3F

OrthoSteveK, 2023-01-17:
Thanks for the very quick response.I just downloaded the latest version of octopod from the AppStore.  It does not show my Octoprint account. I cannot see how to link the app on my phone to my Octoprint account.

OrthoSteveK, 2023-01-17:
Got it sorted, thanks again for your help.
I am getting an error message about web cam, but I do not have a webcam connected.

see screenshot.

[image]

I am not worried and I am not expecting a reply, I’m grateful for your help so far.