gdombiak/OctoPod #637: Login behind Cloudflare Application

brenner-tobias, 2023-03-28:
I am using OctoPod at home and it is awesome. I would also like to use it when not at home, for that I am using a Cloudflare tunnel with 2FA enable (GitHub login), since I do not simply want to open ports and want to have an additional layer of security.

To summarise the set-up (see details [here](https://developers.cloudflare.com/cloudflare-one/applications/configure-apps/)): I am using my domain and Cloudflare as reverse proxy, that has a "Zero Trust" dashboard, where you can create "applications". These applications can be self hosted and be further secured with an additional login method, where I chose GitHub. (this is then all routed via a secure Cloudflared tunnel to my local HomeAssistant, where it is "forwarded" to OctoPrint).

This setup works great for accessing OctoPrint in my web browser: I open the external url, get redirected to the GitHub login and can then access the OctoPrint Login. Unfortunately, the app does not work when entering the external URL.

Now my suggestion is to implement some sort pop-up if there are additional login layers, enabling you to do the login and then use the app.

I am happy to answer any questions you have regarding the suggestion or my set-up.

donslice, 2023-04-07:
I have the same setup - I would go a step further than a login prompt and suggest adding support for Custom Headers. This way, you could configure token authentication in Cloudflare and OctoPod would work seamlessly. 

The app LunaSea implements this and it works flawlessly with Cloudflare Access.

Mathpro, 2023-06-22:
Same here
Would be very interested in this feature

andycarver101, 2023-09-15:
Yes please.  

I have just implemented CF ZT and I am extremely happy with it.  

Is the answer to turn on CORS?

If so then what?

regards,

andycarver101, 2023-09-15:
> I have the same setup - I would go a step further than a login prompt and suggest adding support for Custom Headers. This way, you could configure token authentication in Cloudflare and OctoPod would work seamlessly.
> 
> The app LunaSea implements this and it works flawlessly with Cloudflare Access.

It’s implemented via custom headers for the key and secret of the gateway.

kadaan, 2023-11-21:
Would someone want to get [this commit](https://github.com/gdombiak/OctoPod/commit/b3016baa9f8405e04a93902ef7653c4d342a5571), which adds support for specifying arbitrary headers, into PR for and submitted to this project?  I don't have time.

gdombiak, 2023-11-22:
Thanks @kadaan. I'm happy to take a look and incorporate your changes. Is there a chance you can export your changes or create a PR?

To help me process your chances faster .... could you summarize the data model and the UI changes you made? The diff shows an entire new version of the data model so hard to quickly identify the changes. Was there any other change besides adding 'headers' to the Printer object? 

Thanks,
Gaston

kadaan, 2023-11-22:
I will try over the Xmas holiday.  The summary is that I added a new Headers field that accepts `HeaderKey1=HeaderValue1,...,HeaderKeyN=HeaderValueN`.  It was only added for OctoPrint App Key Connection type.  The underlying camera and rest connections take the value and combine if with the basic auth header.  The TSB and Orbico connections ignore headers. The new data model is because I added a `headers` field.

kadaan, 2023-11-22:
@gdombiak #671

closed 2023-11-27 by gdombiak

andycarver101, 2023-11-29:
So I see this update reached the AppStore! Amazing. I can’t see a change in the GUI though?

gdombiak, 2023-11-30:
Hi @andycarver101,

The new version that is in the App Store does not include this new feature yet. I'm wrapping up a few more things before releasing the next version that will include this feature. Let me know if you would be interested in joining the beta testing team where I will release this feature before making it available to the general public.

Regards,
Gaston

andycarver101, 2023-11-30:
Ok no worries. I won’t be any use to you for beta testing. I’ll stay out of that. Thanks for everyone’s efforts in this.