gdombiak/OctoPod #643: Crashing on M1 machines

scottCheezem, 2023-05-02:
I'm not sure that there is any official support running the iPad version on M1 based Macs, but it is an option to download from the App Store, and I have been enjoying it on desktop as an alternative to the web interface of octopi.  However after the last update, the App crashes almost instantly.  I've enabled the "Share with App developer" under Analytics & Improvements in privacy & security, and I'm happy to share any crash logs or other details I can to help resolve the issue in next update

Pi-Tim, 2023-10-02:
I have this issue too, I know the App’s not officially supported on Mac… but it would be great to use it if anyone’s found a workaround?

drive-c, 2024-07-22:
I too have a crash on my M3 Max MacBook Pro at launch of the app. I've included the crash report below. Please let me know if you want any other information from my end. (I am currently running the public beta for macOS 15.0, but I will submit a crash report for my M2 Max Mac Studio running macOS 14.5 tomorrow.)

Thanks for the great iOS/iPad app!

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Incident Identifier: 44E0B71F-760A-4AB6-818E-EA0AAD1ECA85
CrashReporter Key:   9914179C-C185-7753-561D-E00E81B8BB89
Hardware Model:      Mac15,9
Process:             OctoPod [37358]
Path:                /private/var/folders/*/OctoPod.app/OctoPod
Identifier:          org.OctoPod.OctoPod
Version:             3.28 (2)
AppStoreTools:       15A240a
AppVariant:          1:MacFamily20,1:15
Code Type:           ARM-64 (Native)
Role:                Foreground
Parent Process:      launchd [1]
Coalition:           org.OctoPod.OctoPod [45435]

Date/Time:           2024-07-22 17:16:02.3841 -0600
Launch Time:         2024-07-22 17:16:01.7863 -0600
OS Version:          macOS 15.0 (24A5289h)
Release Type:        User
Report Version:      104

Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Subtype: KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes: 0x0000000000000001, 0x0000000000000000
VM Region Info: 0 is not in any region.  Bytes before following region: 4372004864
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104978000-104a6c000    [  976K] r-x/r-x SM=COW  /var/folders/*/OctoPod.app/OctoPod
Termination Reason: SIGNAL 11 Segmentation fault: 11
Terminating Process: exc handler [37358]

Triggered by Thread:  13

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x1a01cae14 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x1a01dd5b0 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x1a01d39b8 mach_msg_overwrite + 480
3   libsystem_kernel.dylib        	       0x1a01cb15c mach_msg + 24
4   CoreFoundation                	       0x1a02f46cc __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x1a02f2f7c __CFRunLoopRun + 1236
6   CoreFoundation                	       0x1a02f242c CFRunLoopRunSpecific + 572
7   HIToolbox                     	       0x1ab94b13c RunCurrentEventLoopInMode + 292
8   HIToolbox                     	       0x1ab950f2c ReceiveNextEventCommon + 636
9   HIToolbox                     	       0x1ab951090 _BlockUntilNextEventMatchingListInModeWithFilter + 76
10  AppKit                        	       0x1a3e2d7e0 _DPSNextEvent + 660
11  AppKit                        	       0x1a4740774 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
12  AppKit                        	       0x1a3e208ec -[NSApplication run] + 480
13  AppKit                        	       0x1a3df7094 NSApplicationMain + 888
14  AppKit                        	       0x1a4044c08 _NSApplicationMainWithInfoDictionary + 24
15  UIKitMacHelper                	       0x1bb2b7504 UINSApplicationMain + 972
16  UIKitCore                     	       0x1d23cb60c UIApplicationMain + 148
17  OctoPod                       	       0x104a6d82c 0x104978000 + 1005612
18  dyld                          	       0x19fe8b274 start + 2840

Thread 1:
0   libsystem_pthread.dylib       	       0x1a02070e8 start_wqthread + 0

Thread 2::  Dispatch queue: com.apple.CFNetwork.LoaderQ
0   libsystem_trace.dylib         	       0x19ff61170 ___os_trace_formatters_list_block_invoke + 204
1   dyld                          	       0x19fecf3ac DyldSharedCache::forEachImageTextSegment(void (unsigned long long, unsigned long long, unsigned char const*, char const*, bool&) block_pointer) const + 124
2   dyld                          	       0x19febd338 dyld4::APIs::dyld_shared_cache_find_iterate_text(unsigned char const*, char const**, void (dyld_shared_cache_dylib_text_info const*) block_pointer) + 660
[... 585 lines cut ...]
      "rolloutId" : "6434420a89ec2e0a7a38bf5a",
      "factorPackIds" : {

      },
      "deploymentId" : 250000008
    },
    {
      "rolloutId" : "5fb4245a1bbfe8005e33a1e1",
      "factorPackIds" : {

      },
      "deploymentId" : 250000011
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "c47ab4cc-c9c3-4b5d-a87c-e2433ce02597",
      "experimentId" : "6639bc6ba73d460582162323",
      "deploymentId" : 400000006
    }
  ]
}
}

```

gdombiak, 2024-09-22:
Issue #703 was reported today so I tested the app on Mac and fixed the error I was running into. The stacktrace does not look like the one reported in this thread but after testing the app I did not run into other issues. I will close this ticket when next release is out. If you guys still see the app crashing on mac please open a new ticket and include the stack trace as @drive-c did.

closed 2024-09-22 by gdombiak