gdombiak/OctoPod #654: URL is invalid when setting up a printer in the Octopod app

MichaelAbsher, 2023-07-04:
I am trying to add my Ender 3 printer and Raspberrypi into the Octopod app but when I tried to set it up, and I put in the raspberry pi’s URL using the scan feature it gives me the error message. “This URL is invalid.” This is not true, because this is the exact URL I use every time I log into Octoprint on a computer. In a idea how to fix this? I’ve already tried clicking on it, and then immediately clicking off, but that did not work. I also tried on installing the app, but it didn’t work either

gdombiak, 2023-07-04:
Hi @MichaelAbsher,

Could you paste an example of the URL that is being shown as invalid? That will help me test things and take it from there.

Thanks,
Gaston

MichaelAbsher, 2023-07-04:
Here is a picture of it
[image]

gdombiak, 2023-07-04:
Thanks @MichaelAbsher for sharing that screenshot. Could you check if there are whitespaces at the end of the number 206? I tested locally and without a trailing white space the address is valid but the error message appears if I have 1 or more empty spaces at the end.

I can fix for the next release an automatic trim but want to confirm that is your case.

Thanks,
Gaston

MichaelAbsher, 2023-07-05:
I tried what you said but it didn’t work. There is no extra white spaces when I type it in. Still gives me the same error. What should I do now?

3DStyle, 2024-01-13:
I have this issue since day one using the OctoPod app (on iPhone 12 mini). I can add 3D printers only on my other devices (iPads).