gdombiak/OctoPod #656: Multicam not supported

Tezzatron81, 2023-07-26:
Multicam seems not to be supported for the newest version of octoprint. It used to show both my cams but after an upgrade it no longer does. Hopefully an update is coming soon

jim-roberts, 2023-09-08:
I believe this is an issue with the new camera stack. I have an instance without the new camera stack and both cams are available whereas with the new stack only one cam is available.

Tezzatron81, 2023-09-08:
I believe this is correct. It would be nice to have this fixed. But im not holding my breath, and another app is gonna get the same results i fear.

CaptainRedHat, 2023-09-16:
@gdombiak I have a hardware config that doesn't work. I'm running octoprint 1.9.2 on a Raspberry Pi 3. I'm using the RPi Cam version 1 as my main camera and a Microsoft LifeCam USB webcam as my second camera. Both cams work fine in the WebUI. When I added my octoprint server to OctoPod, I noticed that it displayed the Raspi Cam for a second or two before switching to the USB webcam and from that point on, I'm unable to switch between the cameras. What config info are you looking for?

Devirex, 2023-09-29:
Same here, I have the new camera stack serving two different cameras. Both are MJPEG strems on different ports and displaying fine with Safari when going on the direct link. The octopi web interface is also working in safari and displaying both streams with the help of the multicam plugin very smooth. Unfortunately the Octopod app is only showing the latest configured stream in multicam. Adding a third Camera will show two cameras in the Octopod app, so it seems like there is something wrong with the indexing of the streams

Tezzatron81, 2023-10-16:
@Devirex did u actually set up a 3rd camera to use or just as a dummy so to speak?

Tezzatron81, 2023-10-17:
Ok so it seems the app doesn't pick up the default cam, so like @Devirex said, just add another webcam in the multicam pluging with the default camera address. Not an actual physical camera.

jim-roberts, 2024-04-23:
"Adding a third camera" in Multicam " will show two cameras in the octopod app, so it seems like there is something wrong with the indexing of the streams"

I can confirm this gets around the issue. I renamed the 1st instance , which you can't delete, to "Don't Work" :) and cleared the URL fields. Then added another instance (3rd) with the same URL content as the 1st previously contained and Octopod was then able to see both cameras. 

Is this considered a Multicam issue?

h3llb3nt, 2025-01-07:
i have 2 cameras (4 streams: 2 webrtc and 2 mjpeg) setup in multicam in the main interface and only the last 3 in the list will show for me in octopod. no mater which one you put in the first position in multicam it will never show in octopod. none of the webrtc streams work but im aware that this limitation still exists.  i have to essentially double the first stream in mulitcam for all of them to appear in octopod.

SoarVermont, 2026-01-02:
One year later, and I have the exact same issue as h3llb3nt reported above. Fortunately the workaround of doubling one of the cameras in the first Multicam slot also works for me within OctoPod, but then I have an extra camera in the OctoPrint web interface. 

In my case I have four cameras hosted by a separate computer running ustreamer. The Classic Webcam plugin is disabled.