gdombiak/OctoPod #660: Thermal runaway while paused

wkuwake, 2023-10-03:
When paused my extruder temps are set to 0 but only actually cool to 23-28C depending on ambient conditions. To avoid unwanted erroneous messages I have to set the thermal runaway warning to look for a 30 degree C difference between setpoint and reported temperature which for all purposes disables the feature. 

I am aware I can disable this notification or set the difference to 30+. Any other options? 

[image]

GitHubGoody, 2024-02-18:
By paused, do you mean idle (i.e. no print running)? My Prusa MK4 behaves the same way, except I keep getting the thermal runaway warnings even when I set the difference to 25° or even 30°. I think I'll just disable it since Prusas have thermal runaway detection built-in, but curious what @gdombiak thinks about this. It would be nice to be able to use it as a back up.

wkuwake, 2024-02-18:
> By paused, do you mean idle (i.e. no print loaded/running)? Otherwise, why would your printer command 0° during a print?

I mean pausing during an active print. Not idle. Example: filament runout sensor initiated M600 command commands the print to pause, park the head and set nozzle temp to 0 (maintaining bed temp) until the operator returns to fix the filament issue. Also a similar issue during a manually programmed filament change.

Tor-Einar, 2024-05-10:
How to do you disable this warning or change the threshold. I'm not able to find it. And this  warning is very irritating.