gdombiak/OctoPod #680: Cloudflare Service Token not working with new Additional headers in 3.28

bkcberry, 2023-12-07:
I'm accessing OctoPrint externally through a cloudflare ZT application and saw where you guys literally days ago added the ability for octopod to utilize this with service tokens. I created a new token and added it to my CF application, pasted the client id and secret in octopod and hit request, but it's telling me "The data couldn't be read because it isn't in the correct format"

My client id and key were pasted in exactly like this:
`CF-Access-Client-Id: 2ecxxxxxxxxxxxxxxxxxxxxxxxxxxxxx.access, CF-Access-Client-Secret: 1axxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx`

Does anyone have any ideas? Please and thanks

bkcberry, 2023-12-07:
Thanks! This was on the screen to add a new printer

bkcberry, 2023-12-23:
Has anyone ever had a chance to look into this?

AdmiralMichael, 2024-03-24:
I am having this issue as well  

I have my cloudfare tunnel forwarding to NGINX Proxy Manager then to Octoprint (I run pihole and this allows me to keep the traffic local when on the LAN).  So the domain I use resolves to the NGINX IP when on my wifi. 

When I get the error I am on cellular data, when I am on my wifi it works but once I turn off wifi it stops working. 

So it looks like the issue is with the CF tunnel setup.

AdmiralMichael, 2024-03-24:
Nevermind I fixed it, I didn't select Service Auth under Action when creating the application. I had it as Allow.

gdombiak, 2024-03-24:
@AdmiralMichael , good to hear you got it working. Let me know if I should update [the documentation](https://github.com/gdombiak/OctoPod/wiki/Safely-connect-to-OctoPrint-from-outside-of-your-home-with-Cloudflare-Zero-Trust).

Is this still an issue for anyone else?

Thanks,
Gaston

bkcberry, 2024-05-19:
I've just tried this again, verified everything in the instructions, but am getting Invalid HTTP upgrade (302), which is not covered in the instructions