gdombiak/OctoPod #682: OctoPrint-OctoPod not installing

DarrenSchwartz, 2023-12-21:
I'm not able to install the OctoPrint-OctoPod plugin on Octoprint 1.9.3 running on an Orange Pi PC with dietpi (Armbian 6.1.63-current-sunxi). Had no problem installing a different plugin. The last few lines of the plugin log below seem interesting (some talk of successfully uninstalling). I have no idea how to interpret this though. Any help appreciated.

```
Installing plugin "OctoPrint-OctoPod" from https://github.com/gdombiak/OctoPrint-OctoPod/archive/master.zip...
/usr/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmplflhclbm/OctoPrint-OctoPod-master.zip --no-cache-dir --user
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Processing /tmp/tmplflhclbm/OctoPrint-OctoPod-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: OctoPrint in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint-OctoPod==0.3.16) (1.9.3)
Requirement already satisfied: pillow in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint-OctoPod==0.3.16) (10.1.0)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (2021.2.23)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (2021.10.11)
Requirement already satisfied: OctoPrint-PiSupport>=2023.5.24 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (2023.7.31)
Requirement already satisfied: argon2-cffi<22,>=21.3.0 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (21.3.0)
Requirement already satisfied: Babel<2.13,>=2.12.1 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (2.12.1)
Requirement already satisfied: cachelib<0.11,>=0.10.2 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (0.10.2)
Requirement already satisfied: Click<9,>=8.1.3 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (8.1.7)
Requirement already satisfied: colorlog<7,>=6.7.0 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (6.7.0)
Requirement already satisfied: emoji<3,>=2.2.0 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (2.8.0)
Requirement already satisfied: feedparser<7,>=6.0.10 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (6.0.10)
Requirement already satisfied: filetype<2,>=1.2.0 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (1.2.0)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (2.0)
Requirement already satisfied: Flask-Babel<4,>=3.1.0 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (3.1.0)
Requirement already satisfied: Flask-Login<0.7,>=0.6.2 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (0.6.2)
Requirement already satisfied: Flask-Limiter<4,>=3.3.0 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (3.3.1)
Requirement already satisfied: flask<2.3,>=2.2.3 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (2.2.5)
Requirement already satisfied: frozendict<3,>=2.3.7 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (2.3.8)
Requirement already satisfied: future<1,>=0.18.3 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (0.18.3)
Requirement already satisfied: markdown<4,>=3.4.3 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (3.4.4)
Requirement already satisfied: netaddr<0.9,>=0.8 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (0.8.0)
Requirement already satisfied: netifaces<1,>=0.11 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (0.11.0)
Requirement already satisfied: passlib<2,>=1.7.4 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (1.7.4)
Requirement already satisfied: pathvalidate<3,>=2.5.2 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (2.5.2)
Requirement already satisfied: pkginfo<2,>=1.9.6 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (1.9.6)
Requirement already satisfied: psutil<6,>=5.9.4 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (5.9.5)
Requirement already satisfied: pydantic==1.10.12 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (1.10.12)
Requirement already satisfied: pylru<2,>=1.2.1 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (1.2.1)
Requirement already satisfied: pyserial<4,>=3.5 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (3.5)
Requirement already satisfied: PyYAML<7,>=6.0.1 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (6.0.1)
Requirement already satisfied: requests<3,>=2.28.2 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (2.31.0)
Requirement already satisfied: sarge==0.1.7.post1 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (0.1.7.post1)
Requirement already satisfied: semantic-version<3,>=2.10.0 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (2.10.0)
Requirement already satisfied: sentry-sdk<2,>=1.19.1 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (1.29.2)
Requirement already satisfied: tornado<7,>=6.2 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (6.3.3)
Requirement already satisfied: watchdog<3,>=2.3.1 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (2.3.1)
Requirement already satisfied: websocket-client<2,>=1.5.1 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (1.6.1)
Requirement already satisfied: werkzeug<2.3,>=2.2.3 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (2.2.3)
Requirement already satisfied: wrapt<1.16,>=1.15 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (1.15.0)
Requirement already satisfied: zeroconf==0.39.4 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (0.39.4)
Requirement already satisfied: zipstream-ng<2.0.0,>=1.5.0 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (1.6.0)
Requirement already satisfied: blinker<2,>=1.6.1 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (1.6.2)
Requirement already satisfied: class-doc<0.3,>=0.2.6 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages (from OctoPrint->OctoPrint-OctoPod==0.3.16) (0.2.6)
[... cut ...]
  File "/mnt/dietpi_userdata/octoprint/.local/lib/python3.11/site-packages/PIL/Image.py", line 82, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory
2023-12-20 21:31:29,163 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2023.7.31) did not pass check, not loading.
2023-12-20 21:31:29,183 - octoprint.plugins.pluginmanager - WARNING - The plugin was installed successfully, but couldn't be found afterwards to initialize properly during runtime. Please restart OctoPrint.
```

gdombiak, 2023-12-21:
Hi @DarrenSchwartz,

That's interesting. This seems to be an error in the PIL package that is not loading a dependency. Based on information on the internet a quick fix is to install libopenjp2-7 by running this from ssh/command line in your RPi.

> sudo apt install libopenjp2-7

I'm curious about your setup and why it's missing that dependency.

Thanks,
Gaston

DarrenSchwartz, 2023-12-21:
Well thanks! That did the trick! If you care to investigate, here is my Octoprint system info dump. I skimmed them all and I'm not concerned about anything being sensitive. Well aware that some local IP addresses and printing details are in there - no big deal.

[octoprint-systeminfo-20231220214232.zip](https://github.com/gdombiak/OctoPod/files/13746759/octoprint-systeminfo-20231220214232.zip)