gdombiak/OctoPod #683: IOS app won't sync to iCloud - "failed to update some records"

Grumps1969, 2023-12-23:
I have the OctoPod app running on my iPhone 14, but it won't sync printer info to the cloud.
Error message from app's log is:
"_Failed to push changes for printer: http://**[IP on local network]**. Error: Failed to modify some records._"

When I try to run OctoPod on my AppleTV it just shows "retrieving printer info..." and never completes.

I've removed & reinstalled the app on my phone, but it didn't help.
Any idea what setting(s) I might be missing?

Thank you

janfryauf, 2024-01-04:
Exactly the same happening to me. I also tried ‘Reset iCloud data from printers’ and still getting the same error. Apple TV won’t load anything. Any fix?

3DStyle, 2024-01-13:
Same here. My printers will not sync between devices. What's going on?

Grumps1969, 2024-01-13:
Sadly I've given up - never got it working, and at the same time managed to break my printer (damage unrelated to Octoprint).
The new one I've ordered to replace it is not compatible with Octoprint, at least not yet?
Good luck to you all.  Maybe I'll be back some day.

assumebreach, 2024-02-03:
Same issue here!!

corpjester, 2024-02-05:
Same issue here. Can’t sync to iCloud, getting identical error as above.

jatkins11, 2024-02-08:
Sadly I'm getting the same error as well. I did a little googling and found this post on [Stack Exchange](https://stackoverflow.com/questions/51254169/cloudkit-failed-to-modify-some-records-zone-not-found) where there may be an issue with the way the app is trying to access a zone, but I'm not an iOS app developer so I can't say for sure.

camrichmond, 2024-02-12:
Yep, getting the same issue here as well. I was trying to change the name of my printer on my Apple TV and now I can't get it to sync from any iOS devices.

gdombiak, 2024-03-04:
Hi guys,

Sorry this is happening to all of you. Wonder if you can help me gather more information so I can troubleshoot this. Could you do the following and share the results:

1. Open OctoPod and go to "Settings"
2. Click on "Devices"
3. Click on "Reset printers from iCloud"
4. If this fails and click on "View activity log"
5. Paste screenshot or results here

Thanks,
Gaston

billy27607, 2024-03-04:
Click on "Devices"

camrichmond, 2024-03-04:
Sure thing. The first two messages pop up when clicking, "Reset printers from iCloud data."
The last two are from clicking, "Reset iCloud data from printers."

[image]

corpjester, 2024-03-04:
Same result as the others. Nothing happens, screenshot attached:

￼

gdombiak, 2024-03-11:
Hey guys, I created a new version that will help me collect more data to see what’s going on with this error.  If you are already a beta tester go ahead and install the new version and follow the steps I mentioned above. Share a screenshot with the results. You might want to rotate the phone for the screenshot if the text is not fully visible. 

If you are not a beta tester then send me a message and I’ll add you to the list of testers. 

Thanks 
Gaston

camrichmond, 2024-03-11:
> If you are not a beta tester then send me a message and I’ll add you to the list of testers.
I feel stupid for asking, but how do I message you? I'd like to help you out.

billy27607, 2024-03-11:
You can reach me at ***@***.***  I would be happy to help test.  Sent from my iPhone

corpjester, 2024-03-11:
Please add me to the beta testers, I’d be glad to help out. Sent from my iPhone

corpjester, 2024-03-14:
> Hey guys, I created a new version that will help me collect more data to see what’s going on with this error. If you are already a beta tester go ahead and install the new version and follow the [...]

Hi, please add me to the beta list for the version to troubleshoot Octopod. 
Thanks in advance! Hope to hear from you soon.

gdombiak, 2024-03-14:
Good question about sending me a message. :) You can send me an email to gdombiak at gmail.

Thanks,
Gaston

gdombiak, 2024-03-15:
Good news! Thanks Michael Goran for sharing your screenshots and providing vital information. Everyone, give it a try and let me know if things are now resolved.

Thanks,
Gaston

billy27607, 2024-03-15:
I can report that all is working.

gdombiak, 2024-03-15:
Good! I got confirmation from 2 people that this issue is now resolve. Sorry for taking this long to resolve it. Too many other things are keeping me busy atm.

closed 2024-03-15 by gdombiak