gdombiak/OctoPod #684: Progress bar shows wrong percentage

wolflu05, 2024-01-08:
Hello, I found this project a few weeks ago, and really love it to see the progress when I'm not at home, but I noticed, that the progress percentage is completely behind the actually percentage. E.g. when my print takes two hours it shows 50% (that's where I get the push notification) when the print only has about 20 min remaining.

edbrannin, 2025-01-02:
Can confirm: OctoPod and OctoPrint can have very different notions of % complete.

This is with the PrintTimeGenius plugin active:

[image]

[image]

wolflu05, 2025-01-02:
Do you found a solution? 

_(Funny enough I printed something today (haven't printed for a while) and experienced this again.)_

gdombiak, 2025-01-02:
Hey guys,

What plugin (if any) are you using for calculating estimated time? I run [M73 ETA Override](https://plugins.octoprint.org/plugins/m73etaoverride/) and OctoPrint and OctoPod show the same estimates. Estimates are also pretty accurate. 

OctoPod does not calculate estimated time on its own. The information is always coming from OctoPrint's server. The question is what the server is telling OctoPod. :) Some plugins do not change the UI in OctoPod but tell OctoPod something else than what you see. This seems to be your case.

Regards,
Gaston

edbrannin, 2025-01-03:
Hi Gaston,

I'm using [PrintTimeGenius](https://plugins.octoprint.org/plugins/PrintTimeGenius/).

I suspect, but cannot confirm, that OctoPod is showing the OctoPrint-native time-% instead of whatever PrintTimeGenius is setting.

I can tell OctoPod is not showing layer %, because 9/125 have been in the 7-8% range instead of the 11% from my screenshot)

(The thing I'm currently printing has a flat base and then thin walls, and OctoPrint's native estimation seems to do a rolling average that assumes all layers are similar, so for a 6-hour print the estimate quickly goes up to 15-20h and then drops quickly)

wolflu05, 2025-01-19:
I dig a bit deeper into the technical side, here is what I found:

When I do an API request to `GET /api/print`, I get the following for progress:

```json
{
    "completion": 17.799411376478265,
    "filepos": 713339,
    "printTime": 1193,
    "printTimeLeft": 3193,
    "printTimeLeftOrigin": "genius"
}
```

The `completion` value is exactly what I see in octopod. However, octoprint in its webui shows `printTime / (printTime+printTimeLeft)=0,272001824` which I was expecting to see, as this is the "correct" (more accurate) percentage.

gdombiak, 2025-01-19:
Thanks @wolflu05 for that useful information. I just installed PrintTimeGenius and debugged it locally. This plugin is also sending wrong completion information in the websockets connection. In the image below you can see that the browser shows 0% progress while websockets information shows 4.7%.

[image]

gdombiak, 2025-01-19:
Looking at the source code of PrintTimeGenius I also see that they take control on how progress is displayed in the browser. We have 2 options here: 
1. Contact PrintTimeGenius devs and ask to follow the [API specification](https://docs.octoprint.org/en/master/api/datamodel.html#sec-api-datamodel-jobs-progress) or 
2. Hack OctoPod's UI to recognize if this plugin is installed and show progress like  PrintTimeGenius does. 

I will try 1 first and do 2 if we do not get a positive answer.

Thanks for bringing this up and doing the research.
Gaston

gdombiak, 2025-01-19:
I will update OctoPod so that it detects if PrintTimeGenius is installed and if so it will read how it is configured for displaying progress. OctoPod users will have to do no manual configuration to see progress properly.

Stay tuned
Gaston

closed 2025-01-20 by gdombiak

wolflu05, 2025-01-20:
Thank you really much for your quick action. Really appreciate it and looking forward for the new version to be released.

gdombiak, 2025-01-20:
OctoPod iOS has been fixed. While testing it I realized that the OctoPod plugin also needs to be modified to override progress calculation if the PrintTimeGenius plugin is installed. I wished this plugin had not broken the API contract but based on thie github issue you linked I do not sense they are open to fix this. Anyway, working as we speak on fixing the plugin as well and release both .... hopefully today. Apple might take a few days to approve the new release before it is up in the App Store.

Gaston

gdombiak, 2025-01-21:
New release submitted to the App Store. Might take couple of days assuming it will be approved. :)

FYI, new release of OctoPod plugin has been released. Notifications will now report proper progress when PrintTimeGenius plugin is installed

commit a14ce9f2 (2025-01-20): PrintTimeGenius plugin broke API contract for tracking progress. Ugly hack on the client side since this is not going to be resolved on the server. Will also need to do the same for the OctoPod plugin. :(

Closes #684