gdombiak/OctoPod #694: Camera status indicator

Nathan-Julian, 2024-05-26:
It would be really nice if there was some sort of an indicator for when the camera view is live or not. Sometimes when you attempt to move the printer and it doesn't immediately appear to move it is just the camera not updating. Some sort of symbol in the corner to indicate that the camera is not live would be useful.

gdombiak, 2024-05-28:
Hi @Nathan-Julian,

Such a great suggestion. I realized the other day that I needed to give a hint when camera is not actually playing and your suggestion helped me visualize it how to do it. Scheduled for next release.

Thanks,
Gaston

gdombiak, 2024-08-19:
Added a spinner so you can tell when OctoPod is trying to render the camera. I added a shorter timeout (used to be 60 seconds) so you can see the error and ask to retry again (https://github.com/gdombiak/OctoPod/issues/701)

[image]

commit 7d3578e5 (2024-08-19): 1. MJPEG camera now uses a timeout of 15 seconds (not 60) and MJPEGVC and ObicoVC offer the option to reload the camera. Closes #701
2. Added an activity spinner when loading MJPEG Camera. Closes #694
closed 2024-08-19 by gdombiak