gdombiak/OctoPod #696: Error when connecting to OctoEverywhere

Pixelluca, 2024-06-02:
I had OctoPod connected to OctoEverywhere for a couple days, and it's been working fine. However, suddenly everything is blank and the OctoPod app give the following error whenever I open the printer:
> Authentication failed
> incorrect authentication credentials
Checking the access page on both OctoPrint and OctoEverywhere shows that the connections are successful, as they show up in the list. I have tried remaking the connection, but they all result in the same error. I can still connect directly to OctoPrint in OctoPod.

Xlawguardian, 2024-06-02:
I’m having this same issue just started out of now where a few hours ago. I’ve been using the app for several years no problem.

TheAlexClavijo, 2024-06-02:
Same issue here. For years, the app works perfectly. What’s happening?

4ng3l1n0, 2024-06-02:
Same issue here. Seems like something wrong with the connection between Octopod and Octoeverywhere/Octoprind..

dmrowley, 2024-06-02:
Same issue here...

Hedgemebaby, 2024-06-02:
Exactly same issue, nothing changed as have been printing all day then suddenly getting the same error, if you stay on the page after dismissing the initial error it also comes up with HTTP upgrade at the bottom question mark

4ng3l1n0, 2024-06-03:
Works now!

Hedgemebaby, 2024-06-03:
Yep just woke up and everything is back to normal

gdombiak, 2024-06-03:
Good to hear things are back again working. I guess OctoEverywhere had a temporary hiccup.