gdombiak/OctoPod #698: Redesigned Live Activity - thoughts?

tjhorner, 2024-06-14:
Hi there,

First of all, thank you to all the maintainers for the work on this great app! I use it all the time, it's a very invaluable tool.

The Live Activity is very convenient, especially since I have an iPhone with Dynamic Island. But I thought the current design doesn't really fit in with the rest of the system. So I decided to fire up Xcode and take a stab at a new design. Here's what I have so far:

https://github.com/gdombiak/OctoPod/assets/2646487/4bdf2702-dc45-403d-b1ae-40cad564affb

I had two goals in mind when designing it: make it feel more native to iOS and more glanceable. The print progress is now front-and-center, with a large native circular progress view representing it. A live ETA timer is also included (with a nice fancy countdown animation provided by iOS).

I still have a few more views to complete (like the dynamic island popup) but I wanted to get feedback and make sure that a PR for this kind of thing would be acceptable.

Thanks!

tjhorner, 2024-06-14:
Also: while testing the live activity I wanted to stand up my own OctoPod APNs proxy so the data updates properly but it seems that portion is not open source. Could you please release the source for that as well? Thanks!

tjhorner, 2024-06-20:
@gdombiak sorry for the ping - have you had a chance to look at this? I'd love to PR this and get it merged :)

gdombiak, 2024-06-24:
Hi @tjhorner ,

That design looks great. I'm traveling at the moment. I will share with you the source code for the APNS server/proxy. If you want to run your own then you will need your own dev certificates and also install the iOS app using your own dev certs.