gdombiak/OctoPod #701: Feature request: Option to reload camera stream

dapetcu21, 2024-07-19:
The webcam stream can get stuck for quite a lot of reasons:
* Poor connection
* App goes to sleep
* Encoding hitches on the raspberry pi

While ideally some these should be handled and recovered from by the app, a manual reload option would be great (I'm currently killing and restarting the app every time the webcam stream gets stuck)

I'm using OctoPi's new WebRTC webcam stack, if it's relevant.

gdombiak, 2024-08-16:
This is a good idea. I realized that I swipe between cameras to force a refresh since I have multiple cameras which is not evident, optimal or available to everyone. Added to next release.

Thanks,
Gaston

gdombiak, 2024-08-19:
Feature implemented. Testing before pushing new release out. Probably some time this week.

[image]

closed 2024-08-19 by gdombiak

gdombiak, 2024-08-19:
Hi @KayGundhardt,

The refresh button appears when any type of error happens while trying to render the camera. The user will have to manually press the button to retry if they want to give it another shot. Would this work for you?

Regards,
Gaston

KayGundhardt, 2024-08-19:
Sure, that’s a progress :)But as I monitor my farms printers on some tablets running your app it would be great not running all the way from one to the next but them refreshing their viewport themselves when rendering errors occur… Could that maybe be integrated too?

gdombiak, 2024-09-02:
Hi @KayGundhardt, I implemented an exponential backoff algorithm so that OctoPod retries rendering the camera in case of a timeout with an increasing delay between retries. There is a maximum retry of 10 times and delay doubles between retries. That totals to 36 minutes of trying to render the camera before giving up. If the user manually clicks retry, reopens the app, switches camera or printer then retries reset to 0.

commit 7d3578e5 (2024-08-19): 1. MJPEG camera now uses a timeout of 15 seconds (not 60) and MJPEGVC and ObicoVC offer the option to reload the camera. Closes #701
2. Added an activity spinner when loading MJPEG Camera. Closes #694