gdombiak/OctoPod #707: Camera Keeps Frezzeing

ryanfur, 2024-10-25:


rodlord5000, 2024-11-18:
I've also been seeing this - getting a "Request timed out" error. Can reconnect to the stream straight away, but will freeze and time out within about 5 seconds or so

gdombiak, 2024-11-18:
Hi guys,

Thanks for sharing this issue. I would like to be able to reproduce this so would appreciate more information. Could you describe more how you're using the app that lead to this? For instance, I opened the app while I was in the wifi and when I left home with the app still running the video froze when it lost the wifi connection.  Any screenshots would also help. Does the video work when you tap on the refresh icon? 

Thanks,
Gaston

rodlord5000, 2024-11-18:
For me it happens when using the app over wifi, pretty much as soon as I open it. Sometimes it will go for up to 30 seconds before losing the feed but usually around 5s. Other functions of the app are not affected (stats etc. update and controls are responsive). The video does come back after refreshing but disconnects again shortly after.

gdombiak, 2024-11-18:
That @rodlord5000, so to confirm I understood correctly. The video works fine and while watching the camera you then get an error saying that "Request time out"? You then press refresh and works again and then it stops again?

One last question, was this working well in previous versions of the app? 

Thanks,
Gaston

rodlord5000, 2024-11-18:
> Thanks @rodlord5000, so to confirm I understood correctly. The video works fine and while watching the camera you then get an error saying that "Request time out"? You then press refresh and works again and then it stops again?
> 
> One last question, was this working well in previous versions of the app?
> 
> Thanks, Gaston

Yes that's correct, I have only just started using this app last week so I can't say if there was a difference in a previous verison unfortunately.

gdombiak, 2024-11-18:
Thanks @rodlord5000 for this info. I helps a lot to start troubleshooting this. By the way, would it be possible to share a screenshot when you see this error? Maybe there is something else that could help me.

Thanks,
Gaston

rodlord5000, 2024-11-18:
> Thanks @rodlord5000 for this info. I helps a lot to start troubleshooting this. By the way, would it be possible to share a screenshot when you see this error? Maybe there is something else that could help me.
> 
> Thanks, Gaston

Sure thing

[image]
[image]

gdombiak, 2024-11-18:
@rodlord5000, I see that the first screenshot (when things were working well) was in the Panel page and the second screenshot is when you went to the Move page. If you were going to stay in the Panel page, does the error appear after the camera was streaming fine? Second question, if you press the refresh icon does the video start working?

Thanks,
Gaston

rodlord5000, 2024-11-18:
> @rodlord5000, I see that the first screenshot (when things were working well) was in the Panel page and the second screenshot is when you went to the Move page. If you were going to stay in the [...]

Yes it does, I was just moving the plate around to see if it stopped as the refresh prompt can take a while to appear. The video does start working again after a refresh but then soon stops.

gdombiak, 2024-11-18:
@rodlord5000, by any chance are you using a reverse proxy (e.g. nginx) to connect to your OctoPrint instance? If you have one, I wonder how it's configured to support long-lived connections.

rodlord5000, 2024-11-18:
> @rodlord5000, by any chance are you using a reverse proxy (e.g. nginx) to connect to your OctoPrint instance? If you have one, I wonder how it's configured to support long-lived connections.

Yes I use Caddy, the config is as follows:

`octoprint.example.com {
        reverse_proxy octoprint:5000
}`

This has been working well with Cura and the Octoprint web page, but looking at the [documentation](https://community.octoprint.org/t/reverse-proxy-configuration/1107) there is a separate directive for the webcam path - could this be causing the issue?

`octoprint.example.com {
        handle_path /webcam/* {
                reverse_proxy localhost:8080
        }
        reverse_proxy localhost:5000 {
                header_up X-Forwarded-Proto {scheme}
        }
}`

rodlord5000, 2024-11-18:
@gdombiak I've done some testing without using the proxy and it seems to work OK. Will keep an eye on it tonight to see if it keeps working

rodlord5000, 2024-11-19:
> @gdombiak I've done some testing without using the proxy and it seems to work OK. Will keep an eye on it tonight to see if it keeps working

No disconnections while not using the proxy (which is not essential to my setup), so all good here

gdombiak, 2024-11-19:
Thanks @rodlord5000 for the update. It's great to hear that we narrowed it down to the proxy. I think that I might have a solution for this that might help others that do require this proxy. I will see if I can install that reverse proxy (I use ngnix) and see if I can confirm the fix works fine. Worst case, would you be ok testing a beta version to see if it works with your proxy?

Gaston

rodlord5000, 2024-11-19:
> Thanks @rodlord5000 for the update. It's great to hear that we narrowed it down to the proxy. I think that I might have a solution for this that might help others that do require this proxy. I will [...]

Sure thing

gdombiak, 2024-11-19:
@rodlord5000, I installed caddy (via docker) and configured it with something like this:

> :80 {
        handle_path /webcam/* {
                reverse_proxy http://192.162.1.22:8081
        }
        reverse_proxy http://192.162.1.22
}

I'm not able to cause the request timeout error. :( Later in the week I will share a beta version for you to try and confirm wether the fix worked or not.

Thanks again,
Gaston

closed 2024-12-05 by gdombiak

gdombiak, 2024-12-06:
Hi @rodlord5000 , new release is out and I'm curious to hear if that solved the issue when using Caddy. Let me know how it goes if you get a chance to test it.

Thanks,
Gaston

rodlord5000, 2024-12-17:
Hi @gdombiak, can confirm it is working thanks :)

commit c9334b37 (2024-12-05): Indicate to keep http connection alive so webcam keeps streaming. Closes #707