gdombiak/OctoPod #708: iPad Pro m4 camera feed is not working!

mc-ott, 2024-12-04:
I have an iPad mini (5th gen) with ios 18.1.1 and an iPad Pro m4 with ios 18.1.1. The camera feed is not working on the iPad Pro m4 but it works on the iPad mini. I removed and reinstalled Octopod but it did not help, still no video feed, everything else looks ok.

gdombiak, 2024-12-04:
Hi @mc-ott , could you share a screenshot of what you see on the iPad mini? I just tested it on the simulator and this is how the video appears on portrait and landscape modes.

[image]
[image]

Thanks,
Gaston

mc-ott, 2024-12-04:
iPad mini 5th gen works OK, I did not include the photos of the iPad mini. Only my iPad Pro M4 11”, Horizontal view video stream does not work, I just realized Portrait mode works sort of, smaller picture and bottom half of screen is empty. Maybe simulate with 11” screen size also if you already did not do it, photos below…Michel______     ______

Simulator.Screenshot.-.iPad.mini.A17.Pro.-.2024-12-03.at.21.52.17.png (view on web)
Simulator.Screenshot.-.iPad.mini.A17.Pro.-.2024-12-03.at.21.52.59.png (view on web)

mc-ott, 2024-12-05:
iPad mini 5th gen works OK, I did not include the photos of the iPad mini. Only my iPad Pro M4 11”, Horizontal view video stream does not work, I just realized Portrait mode works sort of, smaller picture and bottom half of screen is empty. Maybe simulate with 11” screen size also if you already did not do it, photos below…
[image]
[image]

closed 2024-12-05 by gdombiak

gdombiak, 2024-12-06:
Hi @mc-ott , new release is out. Let me know how things work for you.

Thanks,
Gaston

mc-ott, 2024-12-06:
Works well, identical . to my iPad mini.Thanks,Michel______     ______

commit 8def4907 (2024-12-05): Added support for iPad Pro (11 inch M4) and iPad (13-inch M4). Closes #708